RespiraWorks/pcbreathe #5: tune power supply compensation

inceptionev, 2020-04-15:


inceptionev, 2020-04-16:
#5: retuned DC-DC for 1200khz, lower ripple and better derating on inductor

neelfirst, 2020-04-20:
I made a calculator for the TP54360, maybe it will be useful.

https://docs.google.com/spreadsheets/d/1jEzKwsjBqaTznz5cs5TbF8ZhpgKVj5wWJlGPP70hs7M/edit?usp=sharing

The circuit as is should work.

neelfirst, 2020-04-22:
Using the spreadsheet calculator above. I put in your 1.2MHz frequency and have the following suggestions:
- Change L to 4.7 uH, with ESR <= 25 mohm
- add 22 + 0.1 uF output caps to current 47 uF. Max ESR <= 17 mohm
- change R11 to 7k, C7 to 22n, C6 to 150p

Will check against webench next.

neelfirst, 2020-04-22:
My assumption on the output current was off. Given this correction the values line up better. Something is up with my freq comp calcs, so let's just ignore that and move on.

closed 2020-04-22 by neelfirst