RespiraWorks/pcbreathe #6: complete watchdog implementation

inceptionev, 2020-04-15:


inceptionev, 2020-04-21:
watchdog implemented, 1.6s timeout, with enable

neelfirst, 2020-04-22:
Need some filter caps on WDI and ~WDO before calling this done. 0.1uF should be sufficient.

verityRF, 2020-04-22:
Why do we have the watchdog IC attached to the cycle controller? The STM32 IC already has at least two internal watchdog timers that can reset the silicon internally (without needing to dedicate loop time to some external watchdog). The reset pin on the Nucleo header goes to the reset pin of the STM32 IC (and nothing else), so it's definitely overkill. 

We need it on the Pi instead. As I mentioned before, just put a low resistance power mosfet on the Pi supply pin and it can be controlled either by the cycle controller or this watchdog IC. If using the IC careful that it doesn't reset the Pi until it's done booting (gonna have to be clever with the enable pin). Also if using the IC, get a removable jumper/0R bridge to allow devs to disable the watchdog IC during debugging where the dog might starve and cause headaches.

inceptionev, 2020-04-22:
the Pi also has onboard watchdog timer, in a separate off-chip IC, which is why the hardware watchdog for the Pi was removed.  Do you think the STM32 on-chip watchdog is sufficient?  We can leave the external IC option on the board and just not stuff it if we find that the STM32 internal one does the trick.

inceptionev, 2020-04-22:
The onboard Pi watchdog does have an enable, but there do seem to be some corner cases we have to  watch out for.  I made a comment about this in #9, but forgot to transfer them over to this ticket too.  There some info here on the onboard hardware watchdog built into the BCM2835: https://www.raspberrypi.org/forums/viewtopic.php?t=210974 have a look and see if this seems sufficient for our purposes.

verityRF, 2020-04-22:
From a technical standpoint, there is no gain to have the IC with the STM32. The internal one does not have the shortcomings the Pi watch dog has (same capabilities and configuration of timeout as the IC). Furthermore, it can be disabled automatically by the debugger, unlike the IC version, which would make debugging a PITA. 

I double checked the errata for the device and there are no issues with the WDT. You won't gain any reliability from an external one in my opinion. And again, it's an extra manual thing that the controller has to do rather than using automatically and independently of core functions. It's small enough that we could leave the pads on, but I don't think we should fit it, nor do I think we will end up using it for the controller. 

The Pi one is interesting...sad that it isn't available during boot time because we do have to think about what happens if it does stall during boot. Guess an external one is definitely wanted for that.

inceptionev, 2020-04-23:
I think for the Pi, if it hangs on boot, the WDT isn't really going to save us anyway (assuming the problem is persistent), and we're better off just sounding the alarm.  The discrete WDT has a max timeout of 1.6s, which I think is too short for the Pi bootup, unless we can pet the dog very early in the boot sequence.  I think I'm ok saving this as a beta issue.

verityRF, 2020-04-23:
Aye the Pi WDT is a beta thing. Could close this once Neel's caps are in. I do still recommend putting a small R in series with the output that is NF by default (for dev work builds) so folks aren't pulling hairs when the controller is frozen in debug. That way if somehow we still need that IC folks can just solder the simple resistor in rather than the IC itself.

neelfirst, 2020-04-23:
looks like the latest update is testpoint based access to WDT. This is fine.

I've created #45 as a followup for beta.

closed 2020-04-23 by neelfirst