RespiraWorks/pcbreathe #19: Reverse input diode possible undersized

inceptionev, 2020-04-19:
Rated for 5V, might require more (from Lee)

inceptionev, 2020-04-19:
Also note if motor braking is added to the motor control, it might not be possible to have a diode here.

neelfirst, 2020-04-20:
It's rated for 60V at 5A...

VF for this diode is 0.7V so if we design for Vout = 5 @ 2 A then the input current (assume 9V) is 1.1 A so power dissipated will be 0.77 W, which will raise Tja by 38.5 C. As it can handle 125 C this isn't a concern.

https://www.diodes.com/assets/Datasheets/ds13012.pdf

inceptionev, 2020-04-21:
Removing this diode for Alpha.  I think it's a reasonable expectation for the power to get plugged in correctly.  This diode is going to cause problems anyways if we go to a 4-quadrant motor controller.

closed 2020-04-22 by neelfirst