RespiraWorks/pcbreathe #22: switch to 3.3V scale on ADC inputs

inceptionev, 2020-04-19:


inceptionev, 2020-04-21:
All inputs switched to 3.3V scale.  Resistor values modified to maintain filter values and suitable impedance for ADC sampling

neelfirst, 2020-04-22:
This circuit will do as is, but changing the series R to 1k will give you more voltage range (0-3V instead of 0-2.5V) and a slightly weaker cutoff (-1.2 dB @ 100 Hz vs originally -1.7 dB @ 100 Hz).

If you want the stronger cutoff bump up Cout to 1.8u or 2.2u.

neelfirst, 2020-04-22:
Unrelated - these sensors can only be powered by 5V.

inceptionev, 2020-04-22:
Also keep in mind that BOM changes can be made after the board goes to fab, so we can tune these values once these are on our desks.  If there's a component option that we should add pads for, let me know.

inceptionev, 2020-04-23:
> This circuit will do as is, but changing the series R to 1k will give you more voltage range (0-3V instead of 0-2.5V) and a slightly weaker cutoff (-1.2 dB @ 100 Hz vs originally -1.7 dB @ 100 Hz).
> 
> If you want the stronger cutoff bump up Cout to 1.8u or 2.2u.

That's a typo in the part selection, the lower resistor should be 3K.  Good catch.

inceptionev, 2020-04-23:
corrected/implemented

steve281086, 2020-06-18:
@inceptionev  can we close this issue? Looks like you resolved it.