RespiraWorks/pcbreathe #23: Evaluate voltage sensor source resistance

inceptionev, 2020-04-19:
(from Lee)

inceptionev, 2020-04-21:
Good to check.  The STM32 ADC sampling cap is 5pF, given worst case 3.3V swing and 500hz sampling, the average current draw is 8.25nA, which against a 6.7kohm thevenin source resistance is 55uV, and the instantaneous charge drop against the 0.22uF cap is 75uV, for a total worst case error of 130uV.  The LSB for a the 12bit ADC referenced at 3.3V is 806uV.  The worst case error is less than 1/2 LSB.  We should be good.

inceptionev, 2020-04-21:
Switched R_hi to 35.7k to change to 3.3V scaling.  This changes the thevenin resistance to 7.8k, which represents an increase of 6uV to 64uV, still well below 1/2 LSB when combined with the sampling charge balance.

neelfirst, 2020-04-22:
confirmed

closed 2020-04-22 by neelfirst