RespiraWorks/pcbreathe #25: Replace Nucleo board with STM32 chip

sglow, 2020-04-20:
The Nucleo board really doesn't have much on it that's useful other then the STM32 part.  Why not just place the part down on this board and get rid of the Nucleo board all together?  That would lower cost and put us one step closer to a final design which certainly won't include that board.
The Nucleo board does have a debug interface which is handy, but that can be replaced with a 4 pin connector to allow an external STLINK debugger to be connected to the chip.
Other then the processor and the debugger, the Nucleo board is pretty much just connectors and power supplies.

neelfirst, 2020-04-20:
This is primarily risk mitigation - we would like to sub in the PCB module in the alpha units and test ASAP.

sglow, 2020-04-20:
Ya, I figured that was the thinking.  These parts are pretty dead simple to use, they don't require much more then power in and assuming you're using a QFP package it's probably easier to solder the chip to the board then all those connectors.

neelfirst, 2020-04-22:
Reopening and assigning to Beta, since we would like to get here eventually.