RespiraWorks/pcbreathe #27: Put command pins on separate timer outputs for PWM freq options

inceptionev, 2020-04-21:
(from Lee)

inceptionev, 2020-04-22:
implemented

neelfirst, 2020-04-22:
I'm not sure what this means.

inceptionev, 2020-04-23:
From Stephen Glow  3 days ago
a few  notes:
I'd recommend adding an SPI interface for pressure sensors with at least 4 GPIO lines for selects.  That would leave open the option of using a gauge or absolute pressure sensor for flow measurements.  I've had good results with that and am continuing to test with less expensive components.  It's good to keep our options open there.
For any pin that needs to be a PWM output (like the motor command) you'll want to select a pin that's connected to one of the STM32 timers.  You'll find a table of 'alternate functions' for the various pins.  That's where you can tell which peripherals the pin can be connected to.  Sorry if that's stating the obvious, I'm not sure if you've done much work with the STM32 parts previously.
I'd recommend using a PWM output pin for any pin that needs to drive the solenoid.  Typically solenoids need more power to initially actuate then they need to stay open.  You can deliver full 12 volts to command it to open, the drop down to a lower voltage to hold it open by pwming the output.  It's easier on the solenoid.
You may also want to use a PWM pin for the buzzer.  Depending on the part you select it might require a specific frequency (4kHz is pretty common).
For those timer outputs it's best to select a different timer for different functions.  For example, you'll see pins marked as timer 1 channe 1, timer 1 channel 2, etc.  Selecting pins assigned to different timers allows more flexibility in PWM frequency etc.  It looks like we'll have plenty of options with this part and requirement set, so that shouldn't be an issue.
I've done a lot of work with the STM32 processors, so feel free to DM me when you're selecting pins if anything is unclear.

neelfirst, 2020-04-23:
It looks like the channels are independent even within a timer, so that works for my concern here.

closed 2020-04-23 by neelfirst