RespiraWorks/pcbreathe #31: Hardware serialization for PCBs

neelfirst, 2020-04-22:
This is a production feature.

Followup from #17 - add hardware serial numbers for PCBs.

This could be a traditional R-based board ID, but I've used EEPROM for this as well (non-medical)

inceptionev, 2020-04-22:
There are unique ID ICs we could use, or yeah burn them into an EEPROM

sglow, 2020-05-19:
We can just store it in the flash memory of the STM32, or in the OTP (One Time Programmable) section of flash if we're paranoid

KennethLaws, 2020-05-25:
I think that we can say this is closed from a hardware perspective.  Hardware serialization will be stored in non-volitile memory.  @neelfirst Is this issue tracked by software elsewhere?  If so, I would like to close this here.

neelfirst, 2020-05-26:
Opened https://github.com/RespiraWorks/VentilatorSoftware/issues/376, closing this.

closed 2020-05-26 by neelfirst