RespiraWorks/pcbreathe #32: Non-ideal UART connected to Pi from Nucleo

verityRF, 2020-04-22:
So the RXD/TXD pins that are marked on the Nucleo dev board are link to both the 19x2 header and to the STLink USB debug chip onboard. This means that any communications between the Pi and the Nucleo would interfere with debugging from the PC that would hugely gain from having that serial access over USB from a dev standpoint. Furthermore, any debug messages sent to the Nucleo would interfere with the same link. 

1) Having dev access to the built-in serial pins that are by default connected to the USB debugger is extremely valuable (can print various controller/sensor values over time over serial and have it show up on the host x86 PC for analysis)
2) Having the controller link also shared will cause a lot of headache, especially with introducing potential timing headaches during troubleshooting and debug

I would recommend putting PA2/PA3 (TXD/RXD labeled on the Nucleo) as NC. 

Then use any of the following pins for the Pi link:
MCU Pin Fn Alt Name -> MCU Pin # (pairing by spacing means nothing)

USART3 TX -> PB10
USART3 RX -> PB11

USART3 TX -> PC4
USART3 RX -> PC5

USART3 RX -> PC11
USART3 TX -> PC10

USART4 TX -> PA0
USART4 RX -> PA1
USART4 TX -> PC10 (yes this is not a typo)

USART1 TX -> PA9
USART1 RX -> PA10
USART1 TX -> PB6
USART1 RX -> PB7

You can use any combination as long as they belong to the same USART channel number (must have for instance RX+TX from USART1, or RX+TX from USART3, etc.)

For those unfamiliar with ARM, yes those functions are not shown on the Nucleo or marked on the PCB. All Cortex-M parts have internal configurable pin muxing that allows you to map specific multiple pins to an internal peripheral through software configuration. Yes those pins may overlap/look funny, but they're from the datasheet!

inceptionev, 2020-04-22:
great catch!

inceptionev, 2020-04-22:
Implemented using USART3 with flow control signals

neelfirst, 2020-04-22:
I confirmed this change has happened.

closed 2020-04-22 by neelfirst