RespiraWorks/pcbreathe #33: UART Hardware Flow Control Addition

verityRF, 2020-04-22:
It would be nice if you could route an additional two digital lines between the Pi and the Nucleo for out of band signaling purposes. It'd be nice if you could use the specific RTS/CTS pins (best matches shown for you below). I know the Pi side doesn't have RTS/CTS capabilities, but even if they were attached to general digital pins that's better than nothing. 

If the specific RTS/CTS pins are too hard or cause trouble for rearranging other signals, then any two digital lines would do to anywhere. This just allows us to indicate to both sides of the link if the other side is ready to transmit/receive without relying on the blind serial link.

See #32 - it's similar!

USART3 TX -> PB10
USART3 RX -> PB11
USART3 CTS -> PB13
USART3 RTS -> PB14

USART1 TX -> PA9
USART1 RX -> PA10
USART1 CTS -> PA11
USART1 RTS -> PA12

inceptionev, 2020-04-22:
Implemented using USART3 with flow control signals

neelfirst, 2020-04-22:
Confirmed.

closed 2020-04-22 by neelfirst