RespiraWorks/pcbreathe #34: Triple Check Nucleo Physical Pin Mapping

verityRF, 2020-04-22:
It's kind of hard to tell by the way you did the part definition by making a Nucleo symbol U1 AND the 19x2 header symbols P2/P3 (but NC). I think it would be more straightforward to just have the Nucleo symbol as the two 19x2 headers, at least that's how I would have done it.

Regardless, at first glance, it seems like maybe you have the assignments mirrored/rotated? Again, it's hard to tell, I'm assuming U1A maps to P2, U1B to P3, 1:1 pin number mapping, etc. 

I think a couple people should look at this and maybe have a more clearer representation of the symbol and how it relates to the physical headers.

inceptionev, 2020-04-22:
yeah those are in there to make them appear on the BOM for parts ordering.  What I've done on other designs is to make the module itself just a series of anchoring points, and to have the connectors themselves be the actual connection points, I'll swap that design in, stand by for updates.

inceptionev, 2020-04-22:
*that's how its done for the pi, as well

inceptionev, 2020-04-22:
swap has been made, nucleo board footprint is now mech guide with separately placed connectors (like the Rpi).  Please verify pinouts.

neelfirst, 2020-04-22:
I've verified the pinout based on https://www.st.com/resource/en/schematic_pack/nucleo_64pins_sch.zip

But it would be nice to have a physical unit in hand. I should have one today.

neelfirst, 2020-04-22:
@verityRF if you can provide the triple in check I would be grateful :)

verityRF, 2020-04-22:
Yep - Can do it tonight. Also have the physical one in hand.

verityRF, 2020-04-23:
Seems like you nailed it - orientation and connector assignment matches the Nucleo!

closed 2020-04-23 by verityRF

inceptionev, 2020-04-23:
many thanks @verityRF