RespiraWorks/pcbreathe #37: Use locking power input connector for J1 in later board revisions.

rkevinwatson, 2020-04-22:
For later revisions of the board consider using a locking-type connector instead of a barrel connector for twelve volt power input at J1.

inceptionev, 2020-04-22:
recommend JST PH series for common worldwide availability

inceptionev, 2020-04-22:
recommend JST VH for common worldwide availability

inceptionev, 2020-04-25:
Implemented in alpha since we have the parts, using 3-pin VH connector so as not to be swappable with the motor power connector

inceptionev, 2020-04-25:
ready for review and close

rkevinwatson, 2020-04-25:
Checked schematic and BOM and it's still a barrel connector. Should we pass on this change until next revision?

inceptionev, 2020-04-25:
If you check the Altium files the new connector should be in there.  Once I do an export for mfg, the PDF schematics and gerbers in the NIGHTLY folder will also reflect this.

inceptionev, 2020-04-25:
that's coming in like 10min

rkevinwatson, 2020-04-25:
Okay, cool. I checked power.sch and the connector has been changed, Closing this issue.

closed 2020-04-25 by rkevinwatson