RespiraWorks/pcbreathe #42: Watchdog: suggest making R64 DNP

rkevinwatson, 2020-04-23:
Suggest making R64 at U6 pin 3 a DNP. The data sheet indicates there is an internal pull down resistor and suggests floating the input.

inceptionev, 2020-04-23:
we will probably DNP the whole watchdog circuit in alpha.  Seems like both the PI and the STM32 have onboard or on-chip WDT solutions, and the discrete WDT IC can be considered just a backup

KennethLaws, 2020-05-25:
Would like to go forward with using onboard WDT for both PI and STM32.  If we run into a problem we can re-raise the WDT question but I'd suggest we close this issue. @neelfirst  @inceptionev

KennethLaws, 2020-05-25:
Sight change here.  The comment about R64 is correct as per the data sheet.  The question of removing the WDT all together is addressed in another ticket.  I'm going to remove R64 from the schematic and close this issue.

closed 2020-05-25 by KennethLaws