RespiraWorks/pcbreathe #43: Alarms: do we need an acknowledge button to silence the alarm buzzer?

rkevinwatson, 2020-04-23:
Do we need a button that will acknowledge the alarm condition and silence the buzzer?

neelfirst, 2020-04-23:
Quality of life ++

This raises UI questions though, @rwilbur have y'all discussed hard buttons?

inceptionev, 2020-04-23:
Seems like we can do this on the touchscreen.  The buzzer on this board that is under the Cycle Controller's control is only a backup if the UI is unresponsive, otherwise the UI and its own display and speaker will deliver the alarm along with information.  It will only be activated if the UI computer is unresponsive, in which case a major intervention (i.e. restart the machine or swap it out) is necessary.

neelfirst, 2020-04-23:
I supposed we can power cycle the controller to silence the alarm, but it seems to me this might not be desirable behavior for a beta/prod unit. I'll move the milestone accordingly.

rkevinwatson, 2020-04-23:
Forcing someone to reboot a medical device to silence an alarm would give them a pretty bad impression of the equipment. If you're working in a very high stress environment and the UI on a critical piece of equipment is non-responsive AND making a bunch of noise, you'd probably want to kick it to the curb, or worse, recreate the fax machine scene from Office Space. Maybe consider giving the user some indication that the machine detected a fault and shutdown? Maybe add an LED labeled "Fault Detected" next to the power switch?

neelfirst, 2020-04-23:
We have CTS/RTS hooked up but as general I/O, could leverage that (or heartbeat over UART)

KennethLaws, 2020-05-25:
These see like really good points and questions.  My take is that the touchscreen not working properly is not an option, period.  It has to work 100%.  **Given that assertion, a soft button on the touchscreen to silence the alarm seems a logical solution**.  If there is general agreement on this that would be great and we could close this issue. I guess there is a scenario where you have an issue with the vent but it is keeping somebody alive and there is no replacement.  Then you might want to have a hard switch to disconnect the alarm but is that an edge case worth considering?  It seems in normal practice, if the vent isn't working properly, you would want to replace the vent.