RespiraWorks/pcbreathe #44: consider adding a stepper driver

inceptionev, 2020-04-23:


inceptionev, 2020-04-23:
this will allow us to more prototyping options.

we could possibly use one of the i2c ports to attach an i2c stepper driver such as AMIS 30624

neelfirst, 2020-04-23:
https://www.st.com/en/ecosystems/x-nucleo-ihm03a1.html 

Nucleo shield for stepper motor driver

neelfirst, 2020-04-23:
I documented the pin difference between the Nucleo stepper driver and Edwin's board here.
I created separate tabs that just display the pin conflicts, along with my proposed resolution.
The thrust of the idea is to share the SPI bus across uSD and the stepper, with different chip selects.
Alternatively we could just remove the uSD card when testing the stepper motor.

I'd like a review from @verityRF @inceptionev @rkevinwatson to make sure I didn't go crosseyed making the pin charts. I'm particularly susceptible to MOSI/MISO mixups.

https://docs.google.com/spreadsheets/d/1EP9jQYIz0WdIBkeFmQDyeVXeIfuUcTSwAgnJIIQ7mOM/edit?usp=sharing

neelfirst, 2020-04-24:
@sosaysthecaptain

inceptionev, 2020-04-24:
I don't think switching the solenoid control from CN8p21 to CN8p24 is going to work because CN8p24  is on TIM3, which ties the solenoid PWM to the buzzer frequency.  There is a TIM1 output on this pin but it is inverted.  Not sure if that will be a problem but it would be ideal to use a pin with the positive polarity.

inceptionev, 2020-04-24:
Proposed solution: use CN8p24 (PB1) for CC-ID1 and use CN8p12 (PA12/TIM1CH4) for solenoid control.

inceptionev, 2020-04-24:
Please review the attached schematic to confirm the pin swaps
[20200423-Issue44review-RespiraWorks-Ventilator-DWG-SCH.pdf](https://github.com/RespiraWorks/pcbreathe/files/4526548/20200423-Issue44review-RespiraWorks-Ventilator-DWG-SCH.pdf)

inceptionev, 2020-04-24:
Also please verify, all these pins seem to be on CN10, not CN8

neelfirst, 2020-04-24:
Approved! I updated the spreadsheet linked above, with a new tab 'FINAL' detailing this change.

neelfirst, 2020-04-24:
BTW a clerical error on my part, CN10, not CN8

closed 2020-04-24 by neelfirst