RespiraWorks/pcbreathe #45: Investigate WDT efficacy, necessity

neelfirst, 2020-04-23:
followup from #6 

Do we need a WDT for STM32?
Do we need an external WDT for RPI, given that the onboard WDT is not available during boot? 

@verityRF

verityRF, 2020-04-23:
First half question will be answered once we evaluate the internal WDT on the STM32 and make sure it works as advertised.

inceptionev, 2020-04-24:
on the alpha board I have kept the WDT but disconnected to from both pi and stm32.  I have added testpoints so we can experiment with it if desired.

KennethLaws, 2020-05-25:
I'd like to know the status of evaluating internal WDT on the STM32 or any other feedback on WDT.  If we are going to leave the WDT on board for now, it seems a possible reasoning would be to be ready if it was found to be needed later.  To that end, it would seem logical to bring out input, output, and enable and tie into stm32 pins either hard or with splitpads or zero ohm resistor.

KennethLaws, 2020-05-26:
STM32 has 2x WDT that are clocked independently from the system clock, also a systick timer that can be used to reset the system.