RespiraWorks/pcbreathe #46: Add mounting hardware to BOM

neelfirst, 2020-04-23:
Followup from @toybuilder comment in #7:

"Related to this, I recommend adding BOM-entry components in the schematic to track what screws/standoffs/etc. are needed for the entire electronics sub-assembly so that the finished sub-assembly can be treated as a single FRU (field replaceable unit) in the larger context of the entire machine."

inceptionev, 2020-04-25:
I'll put them on the BOM as DNP so as not to confuse the board house

inceptionev, 2020-04-25:
added parts to the schematic that will show up on BOM, DNP'd for this build so as not to confuse board house sourcing (the computing modules are also DNP'd)

inceptionev, 2020-04-25:
implemented, ready to review for close

KennethLaws, 2020-05-25:
mechanical parts have been added.  This seems sufficient to close the issue

closed 2020-05-25 by KennethLaws