RespiraWorks/pcbreathe #49: Connectors - all R/A

neelfirst, 2020-04-24:
Could help with assembly wiring and harness design to have them all come out at the edge.

inceptionev, 2020-04-24:
We should get @ebhillstrom's thoughts on this, but my thinking was that straight connectors would be better given how crammed the panel is, i.e. not necessarily room to maneuver connectors once the vent is assembled.  Is this a gating issue for alpha?

neelfirst, 2020-04-24:
My bad. I'm having difficulty conjuring the latest mechanical design files. Do we then need a vertical barrel jack?

neelfirst, 2020-04-24:
Per https://respiraworks.slack.com/archives/C011KC7TF46/p1587744941210200 the vote is to keep it your way, @inceptionev. So I guess we're good here.

closed 2020-04-24 by neelfirst

inceptionev, 2020-04-24:
The barrel jack was a convenience item for alpha-as-dev-tool assuming everyone has a 12v 5.5x2.1mm wall adapter at home.  This will be replaced with a locking header in beta (in whatever orientation we settle on)