RespiraWorks/pcbreathe #51: Incorrect Nucleo header numbering on schematic

sglow, 2020-04-24:
A minor point, but the schematic lists the two 2x37 way connectors as CN7 and CN10 on the Nucleo board.  I believe these are actually  CN5 and CN6.

inceptionev, 2020-04-24:
CN5 and CN6 are the single-row arduino-compatible headers.  The 2x19 headers are CN7 and CN10.  I have one in front of me and I checked just to be sure.

sglow, 2020-04-24:
Yup, apparently there are two different Nucleo boards with almost exactly the same names but different connector numbering.  I was looking at the NUCLEO-L452RE-P documentation and the schematic is based on the NUCLEO-L452RE.
The board I bought from Digikey seems to be the -P version which won't work on this board because there are a ton of differences in the pinouts.  :-(

inceptionev, 2020-04-24:
Dangit, well that's unpleasant news.  Digikey doesn't have any of the non-P ones in stock.  All the Nucleo-64 boards are supposed to be at least-footprint compatible (with different pin mux functions of course).  I wonder what happened.

sglow, 2020-04-24:
Ya, it's pretty annoying.  There are a ton of differences in the pinouts, so the -P version won't work.
Looks like they're in stock at Mouser though, so that's good.

inceptionev, 2020-04-24:
Out of the two, the -P version seems more readily available.  If it's just pin differences we can swap pins in the design easily.  I am going to explore this possibility and reopen this ticket.

inceptionev, 2020-04-24:
@sglow do you see any other showstoppers in the -P schematic other than pinout?

sglow, 2020-04-25:
Honestly, I didn't look too closely at it.  The board isn't much more then the STM32 chip, some power supplies and connectors.  I would imagine that either one would work equally well.

sglow, 2020-04-25:
Personally, I wouldn't recommend changing this now.  Mouser has 68 of the non -P version in stock now and we'll only need a few.  Hopefully we'll ditch the Nucleo board and just drop a processor on the next spin pretty soon.

inceptionev, 2020-04-25:
It is more likely that we will use the STM32L452RE rather than the STM32L452RE-P since this is not a low power application, so keeping it the same reduces risk.  The Nucleo boards we have now will just serve as dev acceleration tools for now, and won't be useable on these boards.  At $15 each, I think that's acceptable.  I checked and all the important IO are still accessible, just on different pins.  This was a good catch, thanks @sglow.

inceptionev, 2020-04-25:
Decision made to stay with non-P nucleo boards, which will require purchasing new modules for the alpha PCB.  Closing ticket.

closed 2020-04-25 by inceptionev

neelfirst, 2020-04-25:
FUUUUUUU my module is a -P