RespiraWorks/pcbreathe #52: More flexibility for V-SENSOR-I2C

sglow, 2020-04-24:
There are four connectors which allow remote sensors to be connected via I2C (J2, J6, J7, J8).  Each of these connectors includes a power pin connected to net V-SENSOR-I2C.  That net can carry either 3.3v or 5v power depending on how resistors are populated.
It would be more flexible to allow the power to each of these connectors to be individually selected rather then force the same power level on all four connectors.  This would allow for more flexibility in what remote sensors can be accommodated.
Replicating the zero Ohm resistors R1 and R2 for each I2C connector would accomplish this.

inceptionev, 2020-04-25:
pullups need to be switched too but we can do this

inceptionev, 2020-04-25:
implemented as described

neelfirst, 2020-04-25:
Confirmed

closed 2020-04-25 by neelfirst