RespiraWorks/pcbreathe #54: Need external flash chip on next board

sglow, 2020-04-25:
We can't save data to the STM32 internal flash chip during ventilator operation because doing so could significantly impact the control loop performance.
Adding a small external flash chip is probably the simplest and cheapest solution to this problem.
We don't need a lot of memory.  256kbit would be more then enough to store parameter values.  There are lots of options on digikey for 8-pin SPI flash chips of this size for under $0.50.
It's possible that we won't even need this functionality, but having a spot for an external flash chip on the board would give us the option.

neelfirst, 2020-04-27:
We have an SPI interface for a microSD card for this rev so at least SW dev can proceed unhindered. We can replace with such a chip for prod.

neelfirst, 2020-04-27:
Relates to: https://github.com/RespiraWorks/VentilatorSoftware/issues/56

inceptionev, 2020-05-19:
related to #31