RespiraWorks/pcbreathe #65: Rpi standoffs need to be 2mm taller

inceptionev, 2020-05-14:
@martukas discovered that the Rpi sits at an angle.  Go to a 20mm standoff to avoid flexing the board.

KennethLaws, 2020-05-25:
I don't see any 18mm standoff's on board so, has this been addressed?

inceptionev, 2020-05-25:
They are on the front page of the board, M1/M2/M3/M4.  I'm about to order a few parts from digikey today, so I will order some replacement standoffs to try and will drop a suitable part number here if when I get them and try them out.

KennethLaws, 2020-05-25:
M1/M2/M3/M4 are already 20mm

KennethLaws, 2020-05-25:
the part parameter specifies 20mm

inceptionev, 2020-05-26:
I think the original issue was incorrect.  the existing standoffs were 20mm, and were too short.

inceptionev, 2020-05-26:
updated the original comment

KennethLaws, 2020-05-26:
Great, will wait for updated part number