RespiraWorks/pcbreathe #67: Need to add RTC

neelfirst, 2020-05-19:
RW-SYS-069 states:

The ventilator shall provide non-volatile data storage for at least 5000 events in an event log, including at least the following items: - Ventilation starts and stops - All confirmed Ventilation settings - All confirmed alarm settings - All occurrences and ends of alarms with all their related actions: Inhibitions, cancellations, resets, acknowledge button presses. The ventilator shall provide a date and time stamp for all event log data. The event log shall note any change to the system’s Real Time Clock by logging the Current Date/Time followed by the new Date/Time and a unique event code indicating the change.

Relevant to this ticket: Real Time Clock.

So we need a real time clock.

KennethLaws, 2020-05-26:
the STM32 has RTCwith 375 nA Standby mode, 2.4 uA stop mode

KennethLaws, 2020-05-27:
A battery backup to run the RTC will be needed.  This might also run a power fail alarm to alert in case rails that support controller and computer fail.